team-vigir/flexbe_behavior_engine #140: runtime error: "buffer is too small for requested array" and "Failed to execute state"

stlslm, 2021-02-22:
hi, 

I am having a behavior that runs correctly in one machine, but on another machine, it receives a runtime error when executing a state with:

`[ERROR][time] failed to execute state myState`
`buffer is too small for requested array`
`[ERROR][time] failed to execute state myState`

since I just start to use Flexbe, I haven't got a better insight as to what might cause this error when executing a state.

I failed to trace this error "buffer is too small..." since the error message does not exist anywhere. Any hint would be appreciated  

Thank you.

dcconner, 2023-04-28:
Did you ever track this down more to find the source of the error?  It looks like it would be deep in the code.

If still an issue, I'd add a traceback handling to the course to get more information.

dcconner, 2023-05-17:
Closed without further information

closed 2023-05-17 by dcconner