team-vigir/flexbe_behavior_engine #143: UnicodeDecodeError: 'ascii' codec can't decode byte

simonbogh, 2021-03-25:
When using a special character in the author field for a behavior it crashes during execution. I am pretty sure this used to work.

Here is the output:

```sh
[ERROR] [1616671138.416851, 814.633000] [/behavior_launcher] [rosout]: bad callback: <bound method BehaviorLauncher._callback of <flexbe_widget.behavior_launcher.BehaviorLauncher object at 0x7f2af85ddc10>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/robotlab/catkin_ws/src/flexbe/flexbe_behavior_engine/flexbe_widget/src/flexbe_widget/behavior_launcher.py", line 98, in _callback
    be_selection.behavior_checksum = zlib.adler32(be_content_new.encode()) & 0x7fffffff
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 759: ordinal not in range(128)
```

`echo $LANG` returns `en_GB.UTF-8`

dcconner, 2023-04-28:
Would PR #141  address this issue?

Need to look into this a bit more based on @pschillinger  question there.

dcconner, 2023-06-23:
Not an issue on Python 3 ROS 2 version, and I suspect is OK on Noetic.  If still and issue re-file on the github.com/flexbe site

closed 2023-06-23 by dcconner