team-vigir/flexbe_behavior_engine #147: Highlevel behavior re-use

LoyVanBeek, 2021-05-31:
In my applications, I have a few general high-level behaviors that 'wrap around' a behavior that is application specific. 
For example: a Watchdog behavior keeping an eye on diagnostics, battery etc that includes the actual application behavior as a subbehavior. 

With different applications, I now have this Watchdog copy-pasted in a few places, with a single change: the actual application subbehavior differs. 

Is there any way to accommodate re-use of such behaviors, eg. parametrize this Watchdog with the behavior it 'wraps around'?
I'd really like to be able to set up this Watchdog with the GUI, then eg. specify a state/subbehavior X as exchangeable and then pass a parameter somehow to tell which state/subbehavior Y to switch into the place of X. 

I was able to set up some code to override what state gets imported as a subbehavior and parametrize this via ROS parameters, but that is not supported by the GUI of course. The GUI code is quite complex and I was not able to make relevant changes there unfortunately. Would be a nasty hack anyway. 

I've also looked at behavior synthesis, but there the 'template' is not usable with the GUI (a big reason for using FlexBE).

The proper term for what I'd like to do is Dependency Injection, I think.

pschillinger, 2021-09-11:
Not sure if this is still relevant, but let me still try to answer conceptually. What you describe is indeed not possible and I wouldn't know of a way right now how to easily realize it without a hack. The only thing I would propose to do (without knowing further details) is to turn the Watchdog into an own behavior. Then, for each application behavior, have the behavior itself and a behavior + watchdog version that combines the two with a concurrency container. But not sure how well this fits your use case.

LoyVanBeek, 2021-09-16:
Thanks for the reply! 

In the end we decided to go the simplest route: just copy paste. Not ideal, but the most flexible and understandable for maintenance.