team-vigir/flexbe_behavior_engine #157: Flexbe remains disconnected

Jibrangit, 2021-11-01:
Flexbe was working until yesterday. But today suddenly, FlexBE refuses to connect. I checked out the topics, including /flexbe/heartbeat which seems to be publishing the following message "---" which I guess makes sense as it uses the msg std_msgs/Empty. However, the teminal of flexbe_app throws this error:

"/opt/ros/noetic/lib/python3/dist-packages/rospy/topics/py", line 750 in _invoke_callback cb(msg) File "", line 10, in callback TypeError: load() missing 1 required positional argument: 'Loader'

Can you point me to possible things that could have made this happen suddenly?

airuchen, 2022-05-30:
Hi, 

same problem here. @Lxandr98 @Jibrangit did you find any workaround?

Lxandr98, 2022-05-30:
Yes, I managed to solve my problem. For me the issue lied in the version PyYAML. The version of FlexBE I used needs PyYAML version 3.x. However, my python environment had PyYAML 6.x. So I had to change the version to get FlexBE running.

Maybe this can solve your issue as well :)

airuchen, 2022-05-30:
> Yes, I managed to solve my problem. For me the issue lied in the version PyYAML. The version of FlexBE I used needs PyYAML version 3.x. However, my python environment had PyYAML 6.x. So I had to change the version to get FlexBE running.
> 
> Maybe this can solve your issue as well :)

IT WORKS! Thank you so much. Just curious, how do you debug this?

```bash
pip3 uninstall pyyaml
pip3 install pyyaml==3.* # for me 3.13 works
```

Lxandr98, 2022-05-30:
Good to hear! Honestly, a lot of google and sometimes a bit of luck :) 
I searched for the error message in many different ways and at one point I was lucky and found the right solution.

Jibrangit, 2022-05-30:
> Yes, I managed to solve my problem. For me the issue lied in the version PyYAML. The version of FlexBE I used needs PyYAML version 3.x. However, my python environment had PyYAML 6.x. So I had to change the version to get FlexBE running.
> 
> Maybe this can solve your issue as well :)

Sorry for the late reply and not updating this issue after I found the fix. Yes, that's what I found to be the source of error (pyyaml version) for me back then. I dont have access to the repo since I no longer work there, but I remember downgrading it to 5.x version to make it work. 

I believe this issue can be closed. In short - downgrade your pyyaml to resolve this.

Jibrangit, 2022-05-30:
> > Yes, I managed to solve my problem. For me the issue lied in the version PyYAML. The version of FlexBE I used needs PyYAML version 3.x. However, my python environment had PyYAML 6.x. So I had to change the version to get FlexBE running.
> > Maybe this can solve your issue as well :)
> 
> IT WORKS! Thank you so much. Just curious, how do you debug this?
> 
> quick note:
> 
> ```shell
> pip3 uninstall pyyaml
> pip3 install pyyaml==3.* # for me 3.13 works
> ```

HI, I am not sure what your error is. Could you print out the logs that result from that installation command?

closed 2022-05-30 by Jibrangit