team-vigir/flexbe_behavior_engine #158: Containers do not return output_keys to parent

benmaidel, 2021-11-12:
A Container, Statemachine and Concurrent, do not return the defined output_key to the calling parent.

I've made a little test behavior to reproduce the error:
[image]

[image]

Error message is:
```
ERROR] [1636723819.334961]: Failed to execute state LogData:
No data found for key 'data'
```

workspace to reproduce the error:
[test_ws.tar.gz](https://github.com/team-vigir/flexbe_behavior_engine/files/7527854/test_ws.tar.gz)

benmaidel, 2021-11-25:
I took a closer look at this error. The error only occurs when the parent state machine has not set any userdata keys when executing a container with an output_key. By adding a simple state that sets a random userdata key as the first state, the error no longer occurs.

LoyVanBeek, 2023-02-01:
This is an interesting one. 
I just started looking into this, by putting a breakpoint at https://github.com/mojin-robotics/flexbe_behavior_engine/blob/32e1306/flexbe_core/src/flexbe_core/core/operatable_state_machine.py#L223
It never hits when executing a behavior similar to the one in the screenshot.

joao-aguizo, 2023-02-03:
Defining a key with a default value in the "State Machine Userdata" section of the "Behavior Dashboard" tab corresponding to the output key of the problematic state also seems to work, and is probably a more convenient solution.

LoyVanBeek, 2023-02-03:
Yep, that is the 'workaround' we used. We just expected sub-statemachines/containers to work the same way as states do in this regard and made this mistake a few times (long enough in between to forget, different people). 

I'm currently looking into fixing this and making our expectation work. It seems like the sort of issue that takes a while to find and has a simple fix in the end. I'm currently in the first phase of that though :smile: .

LoyVanBeek, 2023-02-07:
Yep, exactly as expected: simple fix found after a couple days of searching. 
The state machine I used for debugging this can be found in https://gist.github.com/LoyVanBeek/58ae1bb9dfe277467ef1a6383a819a2c, as well as some logs of running it with extended debug `print`s, using the code on this [draft PR](https://github.com/mojin-robotics/flexbe_behavior_engine/pull/24) that I'll extract to a fix PR from in a bit. 

The log was made with this a `ROSCONSOLE_FORMAT='[${node}:${function}:${line}][${severity}]: ${message}'` which excluded the timestamp for better diff-ing.

For debugging, I made something to give each `UserData` instance a unique name so they can be tracked around easily. 

What my expectation was, that returning data from a sub-statemachine worked the exact same as returning stuff from a state: it adds a key to the `UserData`, with remapping etc.
Eg. the nested Container has a state with some output_key `a`, which gets remapped to `b` that the Container listed as it's output_keys and the top-level state machine remaps that to `c`.

[This was not the case](https://gist.github.com/LoyVanBeek/58ae1bb9dfe277467ef1a6383a819a2c#file-userdata_with_container_ud_not_init_2-txt-L163), unless we were using the workaround of course, which is to initialize the userdata key the data is remapped to (`c` in the example above). For simple states rather than nested state machines, this initialization is not needed.

So, what I noticed in the logging, is that in the case the top-level `UserData` key gets initialized, the same `UserData` is being used throughout the execution: the `UserData` named [`C` in this log](https://gist.github.com/LoyVanBeek/58ae1bb9dfe277467ef1a6383a819a2c#file-userdata_with_container_ud_initialized_3-txt-L13)

In case the key is *not* initialized, a completely new set of `UserData` instances gets created, with no reference to the existing ones. This could make sense in and of itself, but not what happens in the initialized case either, so not the intended behavior. 

I tracked this down to [this line in `StateMachine.execute`](https://github.com/mojin-robotics/flexbe_behavior_engine/blob/32e13063056e70bb7c8ea95603505ed40fe3b8aa/flexbe_core/src/flexbe_core/core/state_machine.py#L78)

```python
self._userdata = userdata or UserData()
```

I assume this is meant to create a new `UserData` instance in case the variable `userdata` is `None`. But due to how `UserData.__len__` is implemented, it also evaluates to the 2nd operand in case `userdata` has no data nor references and thus creates a new `UserData` in that case, forgetting any references to the outer scope: the lop-level's `UserData`. 

[`UserData.__len__`](https://github.com/mojin-robotics/flexbe_behavior_engine/blob/32e13063056e70bb7c8ea95603505ed40fe3b8aa/flexbe_core/src/flexbe_core/core/user_data.py#LL71-L72C54) is implemented like so:

```python
    def __len__(self):
        return len(self._data) + len(self._reference)
```

which returns 0 when it has no data and it's reference also has no data. This is the case at hand when the data is uninitialized.
Returning 0 evaluates to `False` and thus the `or` evaluates to the 2nd operand and creates a new empty `UserData`. 

The fix is incredibly simple: do not use the `or` but explicitly check whether `userdata` is `None` and only in that case create a new `UserData`: 

```python
self._userdata = userdata if userdata is not None else UserData()
```

With that fix, the log for the case with the userdata key not initialized looks almost exactly like the initialized case, except for the key not being set at the start of course.

PR for that is coming up.

dcconner, 2023-04-28:
PR #163 merged into develop

closed 2023-04-28 by dcconner